janelia-arduino/Streaming #5: CMake support required

Sterpa, 2020-03-25:
Could you please add CMakeLists.txt to the library to create compatibility with СMake projects?
And also for compatibility with ([ESP-IDF4.0](https://github.com/espressif/esp-idf/tree/release/v4.0) + [arduino-esp32-4.0](https://github.com/espressif/arduino-esp32/tree/idf-release/v4.0) as a component) enviroment combination.

I tried the following CMakeLists.txt code and it performed very well:
```
idf_component_register(SRCS "."
                       INCLUDE_DIRS "src"
                       REQUIRES "arduino")

component_compile_options(-std=c++17)
```

This is a realy great library for arduino users, thanks for its adaptation for github! But this library can also be used for more serious and modern projects on CMake.

peterpolidoro, 2020-05-11:
Hi. Sorry it has taken me so long to respond. If you would like to create a CMakeLists.txt file and submit a pull request I will merge it into the repository. Thanks!

peterpolidoro, 2020-07-17:
I added a CMakeLists.txt file as you suggested. Does this work for you?

closed 2021-01-04 by peterpolidoro