janelia-arduino/Streaming #6: Sometimes the STRING object value is not displayed.

khanmaum, 2020-05-08:
Thank you for this great library.
I use it often because it is easy to understand the code.
But, Recently, there was a phenomenon in which the String value was not printed.

This is  simple test code to print out the String object value.
Especially when there is not enough free memory. It shows well.
I use Arduino Uno. 
and  Streaming - Version :: 6.0.1
```C++
#include <Streaming.h>
 static const char PROGMEM pArray[] =
"1234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789\n"
"2234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789\n"   
"3234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789\n"
"4234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789\n"
"5234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789\n"
"6234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789\n"
"7234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789\n"
"8234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789\n"
"9234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789\n";

void setup() {
 Serial.begin(9600);
 Serial.print("sizeof="); Serial.println(sizeof(pArray));
 String str=(const __FlashStringHelper *)pArray;      // Trans Flash -> RAM
 **Serial << "str=[" << endl << str <<  "]" << endl;**   // Print String value
 Serial.print("str.length()="); Serial.println(str.length());
 Serial.print("availableMemory()="); Serial.println(availableMemory());
}
void loop() {
 }
//----------------------------------------------------------
// this function will return the number of bytes currently free in RAM
// written by David A. Mellis
// based on code by Rob Faludi https://www.faludi.com
int availableMemory() {
  int size = 2048;   // Use 2048 with ATmega328
  byte *buf;

  while ((buf = (byte *) malloc(--size)) == NULL);

  free(buf);
  return size;
}
```

result:  
```
sizeof=901
str=[
]
str.length()=900
availableMemory()=760

```

so If I replace 
```
Serial << "str=[" << endl << str <<  "]" << endl; 
```
with 
```
Serial.println("str=["); Serial.print( str ); Serial.print(']');
```

It is output normally. 
```
sizeof=901
str=[
1234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789
2234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789
3234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789
4234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789
5234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789
6234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789
7234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789
8234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789
9234567890123456789012345678901234567890123456789_1234567890123456789012345678901234567890123456789
]str.length()=900
availableMemory()=762

```

khanmaum, 2020-05-10:
I tried a little more.
When working with STRING, it seems to require approximately twice the STRING capacity.
I think it is because of the work area to attach the values of STRING to each other.

gazoodle, 2020-05-10:
This issue ought to be fixed now. Please can you confirm the fix from my branch to test?

khanmaum, 2020-05-10:
It works well with the changed ver 6.0.2 library.
Thank you for your effort.

closed 2020-05-10 by khanmaum