janelia-arduino/Streaming #8: fatal error: type_traits: No such file or directory

mikrocoder, 2020-07-17:
Hello, your update today is unfortunately faulty. I have a portable IDE 1.8.13 with Arduino Nano Every.

`
In file included from C:\Arduino megaAVR0 Portable\arduino-1.8.13\portable\sketchbook\NanoEvery\NanoEveryPinComplete.ino:11:
C:\Arduino megaAVR0 Portable\arduino-1.8.13\portable\sketchbook\libraries\Streaming\src/Streaming.h:59:10: fatal error: type_traits: No such file or directory
   59 | #include <type_traits>`

peterpolidoro, 2020-07-17:
I removed the CMakeLists.txt, does that fix it?

mikrocoder, 2020-07-17:
Hello, 
sorry no, the same error message.

mikrocoder, 2020-07-17:
Hello,
strangely enough, the installed IDE does not cause any problems. Strange.

mikrocoder, 2020-07-17:
Hello,

I was testing something else. The installed IDE does not cause any problems with an Arduino Mega2560. If I switch to the NanoEvery, the above error message appears again. So the problem is somewhere else.

peterpolidoro, 2020-07-17:
Oh I think it might be failing because it is checking if ARDUINO_ARCH_AVR is defined. I will change that and you can try it again.

mikrocoder, 2020-07-17:
We can try it, although I think that the compilation would have to abort much earlier. But it's worth a try. Thanks already for the live support.   :-)

peterpolidoro, 2020-07-17:
I think I broke it. I will revert it back to before I made any changes today.

mikrocoder, 2020-07-17:
Hello, okay, of course it's not pretty. With v5.0 there was only one unused warning. 

`...\Streaming\src/Streaming.h:102:52: warning: unused parameter 'arg' [-Wunused-parameter]
  102 | inline Print &operator <<(Print &obj, _EndLineCode arg)`

peterpolidoro, 2020-07-17:
Ok, I reset back to the previous commit. Try that out and I will work on the warning.

mikrocoder, 2020-07-17:
Okay, Thanks.

peterpolidoro, 2020-07-17:
Is it working again?

mikrocoder, 2020-07-17:
Hello,

there's been a misunderstanding. I can always go back to v5.0 in the library management. Which I did immediately. Had to do that. I mean, v5.0 works with the warning system.

peterpolidoro, 2020-07-17:
Have you tried version 6.0.2 that I just pushed?

mikrocoder, 2020-07-17:
The v6.0.2 had a bug after all. So I went back to 5.0. Did you rename 5.0 to 6.0.2? It's a little confusing for me right now.

mikrocoder, 2020-07-17:
Just installed the 6.0.2 again > above error "#include <type_traits>"

peterpolidoro, 2020-07-17:
Yes sorry, 6.0.2 has been the version since May, but I had forgotten to tag it. I merged someone else's code for that commit. Let me go back and ask if that person thinks it is ok to remove the type_traits include. Can you use version 5.0 until I get that worked out?

mikrocoder, 2020-07-17:
No problem.   :-)

peterpolidoro, 2020-07-18:
Please try now with version 6.0.3 and see if this fixes it for you.

mikrocoder, 2020-07-18:
Hello, the v6.0.4 works on Mega2560 and Nano Every without warnings and without errors. Flawless. Good job man.   :-) Thanks.

peterpolidoro, 2020-07-18:
Thanks @gazoodle for figuring this out!

closed 2020-07-18 by peterpolidoro

mikrocoder, 2020-07-18:
Why does everyone close a thread right away? Doesn't make any sense. I'd like to know what makes your version different when you take over the lib from gazoodle? Also you have to change the version number in the documentation. Otherwise it's just chaos.

mikrocoder, 2020-07-18:
One more question. If you take the code from gazoodle, what makes your library different? And can you still correct the version number here? The IDE shows 6.0.4. Thanks.

peterpolidoro, 2020-07-19:
Version 6.0.4 should be the one merged with the @gazoodle updates. This version is the one that works for you now, correct?

mikrocoder, 2020-07-19:
I see in the Arduino IDE v6.0.4 and it works correctly. Fine.
But here 
https://github.com/gazoodle/Streaming
and here
https://github.com/janelia-arduino/Streaming
stands v6.0.3
I get confused and wonder where V6.0.4 comes from?