janelia-arduino/Streaming #10: error message below on arduino 1.8.12, streaming library is up to date

dr400, 2020-08-11:
In file included from C:\Users\**\OneDrive\Documents\Arduino\temp_\temp_.ino:5:0:

C:\Users\**\OneDrive\Documents\Arduino\libraries\Streaming\src/Streaming.h:51:10: error: redefinition of 'struct std::is_signed<_Tp>'

   struct is_signed { static const bool value = false; };

gl1024, 2020-08-14:
I encountered the same error, while trying to build the example from here:
https://www.hackster.io/rayburne/esp8266-webserver-using-javascript-and-xml-aa78f1

In file included from C:\Users\***\Downloads\AJAX\AJAX.ino:7:0:

C:\Users\***\Documents\Arduino\libraries\Streaming\src/Streaming.h:51:10: error: redefinition of 'struct std::is_signed<_Tp>'

   struct is_signed { static const bool value = false; };

peterpolidoro, 2020-08-18:
Hi. Which version of the Streaming library are you both using?

gl1024, 2020-08-18:
Hi.  I'm using version 6.0.4.

peterpolidoro, 2020-08-18:
I think I accidentally corrupted the version numbers and that is probably causing this error. I have bumped the latest commit to version 6.0.5. When that becomes available can you please try it and tell me if you encounter the same error? Thanks!

gl1024, 2020-08-18:
Version 6.0.5 is working for me.  Thanks!

closed 2020-08-18 by peterpolidoro