janelia-arduino/Streaming #18: namesspace 'std' please do not use

mikrocoder, 2022-02-24:
Hi,

I'm just trying around with the standard lib C++ and when I include your lib it gives **std** redefine. I have renamed your std to localstd for now. Would you please rename your namespace std? However you like. It's generally better not to use default identifiers. Thank you very much.

peterpolidoro, 2022-02-24:
If I rename the namespace, then it makes it tricky to use the std library in some cases and the custom is_signed implementation in others. I guess I could use an alias for std in the cases where the std library is available, but instead I just changed it to use the custom is_signed implementation for every case. Hopefully that works for all users.

closed 2022-02-24 by peterpolidoro

mikrocoder, 2022-02-25:
Hi, that's just the way it is to get deeper and deeper into programming.  :-)  I think this is the right decision to use the own type traits for all supported controllers. Otherwise there are problems with the namespace. Thanks for the change.