janelia-arduino/Streaming #19: iostream and iomanip not working with Serial2 (or Serial 1, or Serial 3)

peteDDD, 2022-09-14:
I am trying to bring all my C++ code into this decade by using stdlib and STL. I have so far not found a way to get a rich offering of STL features for both ESP32 and AVR processors on Platformio. Perhaps I am just missing something here or the right library. Here is an example of some simple iostreams and iomanip code that works fine with cout but some of which works and some that does not work with Serial2 (for example).

``
#include <Arduino.h>

#include <stdlib.h>  // makes no difference

#include <Streaming.h>

#include <iostream>

    using std::cout;

    using std::right;

#include <iomanip>

    using std::setw;

void setup()

{

  Serial.begin(115200);

  Serial2.begin(9600);

  Serial  << "hello world"   << endl;  // works

  Serial2 << "hello world 2" << endl;  // works

  cout    << right << "hello world" << endl;  // works

  Serial2 << right << "hello world" << endl;  // DOES NOT COMPILE

        // no matching function for call to 'print(std::ios_base& (*&)(std::ios_base&))'

        // invalid conversion from 'std::ios_base& (*)(std::ios_base&)' to 'unsigned char' [-fpermissive]
  

  cout    << setw(16) << "hello world" << endl;  // works

  Serial2 << setw(16) << "hello world" << endl;  // DOES NOT COMPILE

        // no matching function for call to 'Print::print(std::_Setw&)'

  cout    << setw(16) << right << "hello world" << endl;  // works

}

void loop() {}
[/CODE]

peteDDD, 2022-09-29:
Guess no one cares...

peterpolidoro, 2022-09-30:
I never use std in my embedded projects, so I do not know how to help you sorry.

peterpolidoro, 2022-09-30:
Where are you getting the stdlib library?

peterpolidoro, 2022-11-07:
Added a comment about your issue in the documentation.