janelia-arduino/Streaming #23: Library in Platform Registry is Outdated

elon-code, 2023-09-20:
Hello,
The library is not up to date in the platform registry. Please change to tracking github for changes if possible. 

Thanks.

elon-code, 2023-09-21:
Yes, sorry. It says last updated 3 years ago on v1 
[image]

peterpolidoro, 2023-09-21:
I think that is pointing to a different repository. I am not sure how to update that one, perhaps I need to publish this repository to the PlatformIO registry. Do you know how that works?

elon-code, 2023-09-21:
I am not sure, but it seems like there are two ways of doing it. The first way would be to "Claim Ownership" of that repository. There is a link in the right sidebar to do that. https://registry.platformio.org/libraries/mikalhart/Streaming 

The second way would be to publish this repository as a new repository. Here is the info on how to do that. I could probably help with this if this is the route you choose, but I think option 1 is preferable if possible. https://docs.platformio.org/en/latest/librarymanager/creating.html#publishing

If it's a lot of work this is not really necessary. It would be nice to not have the outdated library on the registry (as this could be confusing and could cause people problems), but if it's a lot of work it's not that important to me.

Thanks.