janelia-arduino/Streaming #24: Comma separator for printing multiple variables

avandalen, 2024-08-08:
This is not an issue:
Here is a handy solution that makes it easier to print multiple variables one after the other, using a comma separator. Can this be integrated? I use it a lot..

`#include <Streaming.h>

template<class T> 
inline Print &operator ,(Print &stream, const T arg) 
{ stream.print(" ");
  stream.print(arg); 
  return stream; 
}

int a, b, c, d, e, f;

void setup()
{ Serial.begin(115200);
  Serial << "\nData: " << a, b, c, d, e, f;
  // instead of:
  Serial << "\nData: " << a << " " << b << " " << c << " " << d << " " << e << " " << f <<  endl;
}

void loop()
{  
}`

peterpolidoro, 2024-08-08:
Very clever! I will definitely add this. It may take a week or two before I have time though. Thank you!

peterpolidoro, 2024-09-09:
Added. Thanks!